raspberrypi/noobs #627: Noobs SD cards are a scam

kdichev, 2021-05-13:
I have purchased my 3rd Raspberry pi and every time i had to go back to the shop and buy a new SD card!!! They have never worked for me. Linux, Ubuntu or Windows. What's the point of those SD Cards when they cant even boot???

XECDesign, 2021-05-13:
You'll need to contact the shop you're referring to. We haven't seen any issues with the cards used by official distributors.

kdichev, 2021-05-13:
I have only purchased from the official distributors in 3 countries - Denmark, UK and Germany. How can every time I buy a noobs sd card it does not work?! It does not make sense?!?!

arabovs, 2021-05-13:
I had the same issue from the official UK distributor. It seems like a booting failure

kdichev, 2021-05-13:
@XECDesign can u re-open the issue please?

JamesH65, 2021-05-13:
Has anyone tried reimaging he SD cards with Raspberry Pi Imager to ensure that the cards actually work? ie the SD card if fine, its the content that is bad. I wonder if the NOOBS images are out of date? This can rarely happen as new revisions of boards come out, but the supply chain still has older NOOBS images.

lurch, 2021-05-13:
What model of Raspberry Pi are you using? What version of NOOBS came on the SD cards you bought? (you can check by looking at the `BUILD-DATA` file on a PC). It's possible that some resellers might only be selling very old NOOBS cards, which contain a version of NOOBS that is too old to boot on your Pi?

kdichev, 2021-05-13:
> Has anyone tried reimaging he SD cards with Raspberry Pi Imager to ensure that the cards actually work? ie the SD card if fine, its the content that is bad. I wonder if the NOOBS images are out of [...]

I have tried to re-image the sd cards with no success, had to buy a new sd card. All of my purchases were Raspberry Pi 4s

kdichev, 2021-05-13:
I am just very concerned that I purchase RPi's from 3 different official **_distributors_** and the NOOBS SD cards fail 100% of the time

lurch, 2021-05-13:
> I have tried to re-image the sd cards with no success

Could you give more details about what "with no success" means? Error message from Raspberry Pi Imager, or something else?

kdichev, 2021-05-13:
look, I buy a RPI + noobs sd card (many times) it does not work on first boot (rpi gives an err that it can't read card), then I try to format and install Raspberry Pi Imager and the sd card does not respond, it's like it's not there at all. The only fix after HOURS of work was to BUY a new CARD and I am talking about MULTIPLE PURCHASES. What ever I try to do THESE SD CARDS are DEAD

lurch, 2021-05-14:
Your frustration with things not working is understandable, but it seems unlikely that 3 separately-purchased SD cards would all be "DEAD"? :confused: 

> the sd card does not respond, it's like it's not there at all

Have you tried using a different SD card reader? Sometimes SD card readers have [driver problems](https://github.com/raspberrypi/rpi-imager/issues/94) or are [very unreliable](https://github.com/raspberrypi/rpi-imager/issues/212) ?

kdichev, 2021-05-14:
> Your frustration with things not working is understandable, but it seems unlikely that 3 separately-purchased SD cards would all be "DEAD"? 😕
> 
> > the sd card does not respond, it's like it's not there at all
> 
> Have you tried using a different SD card reader? Sometimes SD card readers have [driver problems](https://github.com/raspberrypi/rpi-imager/issues/94) or are [very unreliable](https://github.com/raspberrypi/rpi-imager/issues/212) ?

yes 3 separate purchases, 2 from https://raspberrypi.dk and one from a local shop https://www.elextra.dk/ (in store purchase) in the span of 8 months, don't remember where i purchased from the UK (will check my history). 

I really doubt that my 4 different machines' sd card slots are broken. Spend yesterday 2 hours trying to "learn" how to format SD cards from terminals on 3 OSes (that fails too) so I just throw the card in the trash and go buy a brand new sd card from a reputable manufacturer.

I really don't want to go an buy another one just to prove my point but as I am saying there is something "fishy" with those SD cards.

lurch, 2021-05-14:
If your cards _are_ genuinely faulty (which seems unlikely to me, but meh :shrug: ) then you should be able to return them (to where you bought them) under warranty, rather than just throwing them in the trash?

> go buy a brand new sd card from a reputable manufacturer.

And these new cards **do** work okay, with none of the problems seen with your "NOOBS" cards?

JamesH65, 2021-05-14:
Note that NOOBS SD cards are completely normal SD cards, just with NOOBS on them instead of blank. To get 3 SD cards that are defective out of the box is completely unheard of. 

Can you take a report what make the SD cards are, or attach a photo of one of the failing ones?

kdichev, 2021-05-14:
> Note that NOOBS SD cards are completely normal SD cards, just with NOOBS on them instead of blank. To get 3 SD cards that are defective out of the box is completely unheard of.
> 
> Can you take a report what make the SD cards are, or attach a photo of one of the failing ones?

Non of the cards were working! Not some!

[image]

What I do:

1. Go to official Rapsberry Pi distributor 
2. Buy a Raspberry Pi and a NOOBS "sd card"
3. Plug Raspberry Pi power supply and "sd card"
4. Raspberry Pi Fails to boot 

I just came back from the shop where I bought a totally new  card and installed my distribution in like 10 sec with no issues, but what's the point of  selling "pre-configured NOOBS SD Card" when they don't work. I am a developer too and get frustrated by this. I can't imagine what other people with no IT skills go thru 

I will repeat myself this happens for the 4th time on my RPi purchases!!!

Am I the most unlucky guy that buys Rpi's months apart and always get a faulty card?! I doubt that!

JamesH65, 2021-05-14:
Firstly, you are the first person to report this, and given we sell a LOT of these things, that is a bit strange - I would have expected  many many reports of completely defective SD cards were it a common problem.

What is difficult to understand is that these cards are actually defective, given you cannot install an OS on them with the Imager.  Can you tell me which distributors you purchased the cards from? I know you are saying official, but which specific ones?

Can you test them with h2testw or similar, to make sure they are not counterfeit?

maxnet, 2021-05-14:
>look, I buy a RPI + noobs sd card (many times) it does not work on first boot (rpi gives an err that it can't read card), then I try to 
>format and install Raspberry Pi Imager and the sd card does not respond, it's like it's not there at all

Pi 4 does say "SD: card detected [some large number]"  at the top, or does it also think nothing is connected?

AubsUK, 2021-06-26:
FYI... I've just added a solution (at least for me) to the thread I started at https://github.com/raspberrypi/rpi-imager/issues/94 

This is a built-in card reader in a Dell laptop.

closed 2022-01-13 by XECDesign