raspberrypi/noobs #628: Add, just released, Lakka 3.0 to OS list ?

MopheusDG, 2021-05-22:
Hi, Lakka just launched version 3.0 and it's based on LE 9.2 and 64 bits now, so my questions/doubts are:

How can we request an OS to be added/updated to the main OSes list ?

Thanks.

XECDesign, 2021-05-26:
I'd just need the appropriate snippet to go into https://downloads.raspberrypi.org/os_list_v3.json

procount, 2021-05-26:
@XECDesign - I don't think the current NOOBS files are available for Lakka - they return 404.
Their website states that multibooting is not a priority at the moment and the last noobs image was from 2 months ago.

XECDesign, 2021-05-26:
Oh, that's not great. The expectation is that maintainers will keep an eye on their distro, keep it up to date and give a heads up if the json file requires changes. Happy to add Lakka back when it becomes a priority.

coldspark29, 2021-05-26:
@XECDesign I notifed the Lakka team and you will be notified in this thread when the noobs files are up again

ToKe79, 2021-05-26:
Hi @XECDesign,
I have uploaded the new release of Lakka for NOOBS to our site: https://le.builds.lakka.tv/noobs/
Is it possible to update NOOBS json list, or is there more to do from our side?
Thanks.

procount, 2021-05-26:
Hi @ToKe79,
You just need to provide @XECDesign with an updated fragment for the os_list_v3.json file.
I have included the previous one below that you need to update with information from the respective os.json and partitions.json files from each new version and add a new fragment for any new version (like the 64 bit version).

```
        {
            "description": "The DIY retro emulation console",
            "download_size": 367904604,
            "group": "Games",
            "icon": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/Lakka_RPi4.png",
            "marketing_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/marketing.tar",
            "nominal_size": 1024,
            "os_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/os.json",
            "os_name": "Lakka_RPi4",
            "partition_setup": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/partition_setup.sh",
            "partitions_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/partitions.json",
            "release_date": "2020-01-21",
            "supported_models": [
                "Pi 4"
            ],
            "supports_backup": true,
            "tarballs": [
                "http://le.builds.lakka.tv/noobs/Lakka_RPi4/System.tar.xz",
                "http://le.builds.lakka.tv/noobs/Lakka_RPi4/Storage.tar.xz"
            ],
            "url": "http://www.lakka.tv/"
        },
        {
            "description": "The DIY retro emulation console",
            "download_size": 347310680,
            "group": "Games",
            "icon": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/Lakka_RPi2.png",
            "marketing_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/marketing.tar",
            "nominal_size": 1024,
            "os_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/os.json",
            "os_name": "Lakka_RPi2",
            "partition_setup": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/partition_setup.sh",
            "partitions_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/partitions.json",
            "release_date": "2020-01-21",
            "supported_models": [
                "Pi 2",
                "Pi 3"
            ],
            "supports_backup": true,
            "tarballs": [
                "http://le.builds.lakka.tv/noobs/Lakka_RPi2/System.tar.xz",
                "http://le.builds.lakka.tv/noobs/Lakka_RPi2/Storage.tar.xz"
            ],
            "url": "http://www.lakka.tv/"
        },
        {
            "description": "The DIY retro emulation console",
            "download_size": 297833320,
            "group": "Games",
            "icon": "http://le.builds.lakka.tv/noobs/Lakka_RPi/Lakka_RPi.png",
            "marketing_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi/marketing.tar",
            "nominal_size": 1024,
            "os_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi/os.json",
            "os_name": "Lakka_RPi",
            "partition_setup": "http://le.builds.lakka.tv/noobs/Lakka_RPi/partition_setup.sh",
            "partitions_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi/partitions.json",
            "release_date": "2020-01-21",
            "supported_models": [
                "Pi Model",
                "Pi Compute Module",
                "Pi Zero"
            ],
            "supports_backup": true,
            "tarballs": [
                "http://le.builds.lakka.tv/noobs/Lakka_RPi/System.tar.xz",
                "http://le.builds.lakka.tv/noobs/Lakka_RPi/Storage.tar.xz"
            ],
            "url": "http://www.lakka.tv/"
        },
```

ToKe79, 2021-05-26:
Thanks @procount, the fragmets did not change, except the sizes and release dates.
They can be obtained from:
https://le.builds.lakka.tv/noobs/Lakka_RPi/os.json
https://le.builds.lakka.tv/noobs/Lakka_RPi2/os.json
https://le.builds.lakka.tv/noobs/Lakka_RPi4/os.json
https://le.builds.lakka.tv/noobs/Lakka_RPi4-64/os.json
Does this help?

ToKe79, 2021-05-26:
I see now, so those are not copy/paste the os.json, but also links to other stuff, like marketing.tar and partition info/script - but those remained same...

```
        {
            "description": "The DIY retro emulation console",
            "download_size": 438440324,
            "group": "Games",
            "icon": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/Lakka_RPi4.png",
            "marketing_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/marketing.tar",
            "nominal_size": 1024,
            "os_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/os.json",
            "os_name": "Lakka_RPi4",
            "partition_setup": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/partition_setup.sh",
            "partitions_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/partitions.json",
            "release_date": "2021-05-22",
            "supported_models": [
                "Pi 4"
            ],
            "supports_backup": true,
            "tarballs": [
                "http://le.builds.lakka.tv/noobs/Lakka_RPi4/System.tar.xz",
                "http://le.builds.lakka.tv/noobs/Lakka_RPi4/Storage.tar.xz"
            ],
            "url": "http://www.lakka.tv/"
        },
        {
            "description": "The DIY retro emulation console",
            "download_size": 416884828,
            "group": "Games",
            "icon": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/Lakka_RPi2.png",
            "marketing_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/marketing.tar",
            "nominal_size": 1024,
            "os_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/os.json",
            "os_name": "Lakka_RPi2",
            "partition_setup": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/partition_setup.sh",
            "partitions_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/partitions.json",
            "release_date": "2021-05-22",
            "supported_models": [
                "Pi 2",
                "Pi 3"
            ],
            "supports_backup": true,
            "tarballs": [
                "http://le.builds.lakka.tv/noobs/Lakka_RPi2/System.tar.xz",
                "http://le.builds.lakka.tv/noobs/Lakka_RPi2/Storage.tar.xz"
            ],
            "url": "http://www.lakka.tv/"
        },
        {
            "description": "The DIY retro emulation console",
            "download_size": 335700104,
            "group": "Games",
            "icon": "http://le.builds.lakka.tv/noobs/Lakka_RPi/Lakka_RPi.png",
            "marketing_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi/marketing.tar",
            "nominal_size": 1024,
            "os_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi/os.json",
            "os_name": "Lakka_RPi",
            "partition_setup": "http://le.builds.lakka.tv/noobs/Lakka_RPi/partition_setup.sh",
            "partitions_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi/partitions.json",
            "release_date": "2021-05-22",
            "supported_models": [
                "Pi Model",
                "Pi Compute Module",
                "Pi Zero"
            ],
            "supports_backup": true,
            "tarballs": [
                "http://le.builds.lakka.tv/noobs/Lakka_RPi/System.tar.xz",
                "http://le.builds.lakka.tv/noobs/Lakka_RPi/Storage.tar.xz"
            ],
            "url": "http://www.lakka.tv/"
        },
```

There is also aarch64 version:

```
            "description": "The DIY retro emulation console",
            "download_size": 443116980,
            "group": "Games",
            "icon": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/Lakka_RPi4.png",
            "marketing_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/marketing.tar",
            "nominal_size": 1024,
            "os_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/os.json",
            "os_name": "Lakka_RPi4_64",
            "partition_setup": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/partition_setup.sh",
            "partitions_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/partitions.json",
            "release_date": "2021-05-22",
            "supported_models": [
                "Pi 4"
            ],
            "supports_backup": true,
            "tarballs": [
                "http://le.builds.lakka.tv/noobs/Lakka_RPi4_64/System.tar.xz",
                "http://le.builds.lakka.tv/noobs/Lakka_RPi4_64/Storage.tar.xz"
            ],
            "url": "http://www.lakka.tv/"
        },
```

XECDesign, 2021-05-27:
Many thanks. It should be there now.

coldspark29, 2021-05-27:
Works likea charm! :)

closed 2021-05-27 by lurch