raspberrypi/noobs #633: LibreELEC: cmdline.txt boot based on disk LABEL is too weak

efa, 2021-08-15:
### Describe the bug
the boot based on disk LABELS is too weak
changing the labels results in LE unbootable

### To Reproduce
Steps to reproduce the behavior:
1. in NOOBS install LibreELEC
2. try if it can boot
3. shut it down
4. change the disk label of LibreELEC SYSTEM partition
5. try if it can boot: NO, see log

### Informations
 - LE Version: v9.2.6
 - Hardware Platform: Raspberry Pi3B+

### Log file
the error on boot is:

```
*** Error in mount_flash: mount_common: Could not mount LABEL=System ***
### Starting debugging shell for boot step: mount_flash... type  exit  to quit ###
#
```

### Additional context
I tryed to set back the disk label partitions with:
```
$ sudo mlabel -v  -i /dev/mmcblk0p8 ::System
$ sudo e2label /dev/mmcblk0p9 Storage
```
but the error remian the same.
I made a try with:
```
$ sudo mlabel -v  -i /dev/mmcblk0p8 ::SYSTEM
```
just to be sure this is not related to short/long name on FAT32 disk label, but got same error.

Then I also tryed to set 'cmdline.txt' to:
```
boot=UUID=3E30-E976 disk=UUID=21a3db18-287e-4e80-8645-e0ec29971ffe debugging
```
where UUID are the one for System and Storage partitions, but then got the error log below:
```
Framebuffer vertical res: 1050
Framebuffer spalsh image: /splash/splash-1080.png
mount: mounting /flash/SYSTEM in /sysroot failed: No such file or directory
... (total 15 times)
mount: mounting /flash/SYSTEM in /sysroot failed: No such file or directory
*** Error in mount_sysroot: mount_common: Could not mount /flash/SYSTEM ***
### Starting debugging shell for boot step: mount_sysroot... type  exit  to quit ###
#
```
Should be somewhere something that continue to look for SYSTEM also with UUID set, and do not find the label also once restored to 'System'
The only thing I could do to solve the trouble was a reinstallation of LE.
I made a new try, but same issue, so it is reproducible.
As now I reinstalled LE and keep its partition labels as default, but it is very confusing on an uSD with other OS.

I noticed on another old installation of LE for Raspberry PI1B, the cmdline.txt was already based on UUID from the beginning and infact I changed the disk labels many time without any trouble.

lurch, 2021-08-15:
I think this sounds like a "problem" with LibreELEC (and the way it chooses to integrate with NOOBS), rather than a problem with NOOBS itself?

Obviously the "workaround" is to not go changing the disk labels!

efa, 2021-08-15:
seems NOOBS v3.7 still install LibreELEC with 'cmdline.txt' that use partition labels.
I already tried to open an issue on LibreELEC, but as now I'm not sure if the file:
'config/noobs/partition_setup.sh' that create the cmdline.txt file for LE is in charge to NOOBS or LibreELEC?

lurch, 2021-08-16:
NOOBS _calls_ `partition_setup.sh` (see [here](https://github.com/raspberrypi/noobs/blob/master/recovery/multiimagewritethread.cpp#L568)) but the _contents_ of `partition_setup.sh` is entirely the responsibility of each OS (so in this case, LibreELEC).
Of course, that's only my understanding!

efa, 2021-08-16:
from the code of 'multiimagewritethread.cpp', if the partition label is set, it set id with the label. Only when the label is empty, set id with the UUID.
To make the mechanism less weak, it would be better to always use the UUID.

lurch, 2021-08-17:
This is kinda going beyond my current knowledge, so perhaps @procount can provide some useful information?
(also, don't forget that NOOBS needs to maintain backwards-compatibility with all the OSes it supports - it can't wilfully change the parameters that get passed to `partition_setup.sh` (and risk breaking other OSes) just because one user wants to be able to manually rename their partition labels :laughing: )

efa, 2021-08-17:
I understand, breaking the interface is forbidden, but depens on the meaning on the 'id', it seems the uuid more than a label.
Anyway, what say the documentation of that interface between NOOBS and all the OSs?

procount, 2021-08-22:
> perhaps @procount can provide some useful information?

Well, this is quite complicated and very much dependent on the operating system being installed.
In general, most Linux distributions have the ability to locate and mount their partitions via various means:

- device e.g. /dev/mmcblk0p1 or /dev/sda1
- partuuid e.g. PARTUUID=1232ABCD-01
- label e.g. LABEL=System
- UUID e.g. UUID=(some long hex string)
- etc. (and I'm totally ignoring UEFI and GPT systems)

Most OSes comprise 2 partitions: a FAT32 boot partition and an ext4 rootfs partition. Under normal installations, the boot partition will be on the first partition, and the root partition will be on the second. But when you want to boot several OSes from the same device, this layout cannot be maintained and the partition references must be updated by the installer (i.e. NOOBS).

IIRC, the early versions of NOOBS only used device references such as /dev/mmcblk0p2 because OSes could only really be booted off the SD card and this had a fixed reference. An exception was made for OSes that required mounting partitions by label. When USB booting became possible, device references were not so convenient because multiple USB drives could be installed and their references may change depending on which was detected first. A more permanent way of referencing the USB drive was required, so PARTUUID was used for USB installations. 

Despite NOOBS having these preferences for partition references, it also depends on whether the OS can support them. For example, not all OSes support PARTUUID format, so these OSes are excluded from USB installations. 
Using disk labels for partitition references is an alternative method to PARTUUIDS to make the installation independent of the device reference. Ubuntu is one such OS that uses disk references. Although it can support other partition references, if it is ever upgraded it will revert to disk labels again. Therefore disk labels need to be used to avoid it breaking after an upgrade.
LibreELEC is an OS that also uses disk labels and expects its partitions to have those specific labels. Changing them will break the installation.
Since disk labels are required to be unique, using specifc disk lable for partiion references also means only one copy of such an OS can be installed in a multi-boot environment.

PINN supports a couple of extra options that permit OSes that cannot use PARTUUIDS to be installed using device references (with the obvious proviso that only one USB device should be installed at boot to guarantee it will boot properly), and additional checks for those OSes that require partition labels to be used.

Remember that most OSes are not designed to be used in a multi-boot environment, so the fact that NOOBS and PINN can install so many of them in a multi-boot environment is quite amazing. It is not a given that every OS can be installed in this way. Some have limitations and others just cannot be installed at all.

lurch, 2021-08-22:
Thanks for those extra details @procount , that kinda reinforces my earlier statement (and suspicion) that this is more a problem with the OS (i.e. LibreELEC) than it is with NOOBS or PINN.
(BTW would you mind updating your comment to use "partition labels" instead of "disk labels" - where/if it makes sense to do so?)

procount, 2021-08-22:
Done. 😄

closed 2021-08-31 by XECDesign