raspberrypi/noobs #634: NOOBS links disappeared from raspberrypi.org

efa, 2021-08-15:
In Software page: https://www.raspberrypi.org/software/ there are no links or words to NOOBS, only Raspberry Pi Imager.

Even many pages on the site still give instructions how to use NOOBS, but it is not easy to find the current download link:
https://downloads.raspberrypi.org/NOOBS/

JamesH65, 2021-08-18:
Imager is now our recommended way of installing an OS on he SD card, NOOBS is effectively deprecated. We will be removing references to it through the documentation over time. 

@aallan Removing NOOBS shoul go on some docs task list somewhere.

lurch, 2021-08-18:
@aallan In case it's helpful:
```
$ git grep -i noobs
documentation/asciidoc/computers/configuration/localisation.adoc:If you've installed Raspberry Pi OS using NOOBS, it should automatically pick up the same language you were using within NOOBS. But if you want to select a different language, or if you've installed Raspberry Pi OS from a standalone image, use xref:configuration.adoc#change-locale[raspi-config].
documentation/asciidoc/computers/configuration/localisation.adoc:If you've installed Raspberry Pi OS using NOOBS, it should automatically pick up the same keyboard you were using in NOOBS. But if you want to select a different keyboard, or if you've installed Raspberry Pi OS from a standalone image, use xref:configuration.adoc#change-keyboard-layout[raspi-config].
documentation/asciidoc/computers/configuration/raspi-config.adoc:If you have installed Raspberry Pi OS using NOOBS, the filesystem will have been expanded automatically. There may be a rare occasion where this is not the case, e.g. if you have copied a smaller SD card onto a larger one. In this case, you should use this option to expand your installation to fill the whole SD card, giving you more space to use for files. You will need to reboot the Raspberry Pi to make this available. Note that there is no confirmation: selecting the option begins the partition expansion immediately.
documentation/asciidoc/computers/linux_kernel/building.adoc:If it's a NOOBS card, you should see something like this:
documentation/asciidoc/computers/linux_kernel/building.adoc:Mount these first, adjusting the partition numbers for NOOBS cards (as necessary):
```

@efa Are there any obvious places on the Raspberry Pi website that still recommend / document NOOBS ?

aallan, 2021-08-18:
> Removing NOOBS should go on some docs task list somewhere.

It is. Considering how integrated into the docs it was, it's mostly gone. I'll get rid of those references @lurch found and that'll be that.

efa, 2021-08-18:
i googled and report here only the first two links:
https://projects.raspberrypi.org/en/projects/noobs-install
https://www.raspberrypi.org/blog/introducing-noobs/

BTW: I think most of the users used to install more than one OS with NOOBS, if you keep out multiboot from Imager features, you are forcing users to  use PINN or BerryBoot. Is this wanted?

JamesH65, 2021-08-18:
We are happy for Berryboot or PINN to take up the slack in the multiboot area.

efa, 2021-08-18:
in the reprecation banner at:
https://www.raspberrypi.org/documentation/computers/getting-started.html
you should link the download link for the last version available

Note that NOOBS can be "installed" on every OS (as it do not require installation).
As now Imager work only on Windows, macOS and Ubuntu Linux.
So you are cutting out all Linux distributions. I'm using Debian that is the most similar to Ubuntu (or the inverse), and do not install. RedHat and rpm derivatives are out.
Please generate a cross-distro package like AppImage:
https://linuxhint.com/snap_vs_flatpak_vs_appimage/

maxnet, 2021-08-18:
>So you are cutting out all Linux distributions.

You can compile Imager from source on pretty much any recent Linux distribution.

efa, 2021-08-18:
Most users are not developers also on Linux, so do know how to compile Imager or another application

lurch, 2021-08-18:
> https://www.raspberrypi.org/blog/introducing-noobs/

That's a blog post from 2013???

> Please generate a cross-distro package like AppImage:

A third-party user has created a [snap package](https://snapcraft.io/rpi-imager) and I guess there's no reason that other third-parties couldn't do the same for other packaging formats?

maxnet, 2021-08-18:
>A third-party user has created a snap package and I guess there's no reason that other third-parties couldn't do the same for 
>other packaging formats?

Not sure if the snap is such a good example. Given the major issues it has.
E.g. failing to start on Linux distributions that use wayland, and handling of .zip files that should be extracted to the FAT32 partition (e.g. "eeprom recovery") option being broken.

https://github.com/popey/imager-snap/issues/8
https://github.com/popey/imager-snap/issues/13
https://github.com/popey/imager-snap/issues/10

Would be better if the Linux distributions included native packages, than having too much faith in container formats.
Recall Fedora, ArchLinux, Gentoo and RPI OS do have rpi-imager in their own repository.

lurch, 2021-08-19:
> Recall Fedora, ArchLinux, Gentoo and RPI OS do have rpi-imager in their own repository.

Ahh nice. Is it maybe worth mentioning that in https://github.com/raspberrypi/rpi-imager/blob/qml/README.md ?

efa, 2021-08-19:
I made a try and follow the build instructions on github on my Debian and work. I found the Imager icons in the Accessories menù and wit start.

Note: the debuild comand create many files in the directories where I run git. I kept only 'rpi-imager_1.6.2_amd64.deb', but I imagine many users get confused.

Can you please link the github page on the main download page: https://www.raspberrypi.org/software/ ?
Something like: on other Linux distribution, try to build yourself following instruction at: ...
thank you

efa, 2021-08-19:
snap require lot of disk space for base system, install a daemon, and require custom commands to manage upgrade/remove, I think most users do not want such. So in the available portable packager list is not good

aallan, 2021-08-20:
> in the deprecation banner at: https://www.raspberrypi.org/documentation/computers/getting-started.html you should link the download link for the last version available

Nope. We aren't recommending NOOBS to new users. We therefore don't want them downloading it.

efa, 2021-08-20:
ok for new users, but I think others have the right to find the link to the last version released before deprecation.
With all the other open programs that have been abandoned I have always seen indicating where to download the last available.

aallan, 2021-08-20:
> ok for new users, but I think others have the right to find the link to the last version released before deprecation.
> 
> With all the other open programs that have been abandoned I have always seen indicating where to download the last available. 

You're writing in an issue on the noobs repo. Look under releases? 

I will not be adding links to NOOBS from the documentation.

lurch, 2021-08-20:
> the noobs repo. Look under releases?

On a side-note (as if this issue didn't already have enough tangents!) why does https://github.com/raspberrypi/noobs/releases still say that the latest release is v3.0 @XECDesign  ?

efa, 2021-08-20:
once on github repo OK anyone can find that.
I asked to add a link to github in the deprecation banner at: https://www.raspberrypi.org/documentation/computers/getting-started.html

aallan, 2021-08-21:
> I asked to add a link to github in the deprecation banner at: https://www.raspberrypi.org/documentation/computers/getting-started.html

Sorry, I don't think that's an appropriate addition. I won't be added a link back to the NOOBS repo from the documentation site.

Dewidos, 2021-08-24:
But please, just don't remove the download mirror and the repo. Let it be deprecated, but don't let it be totally gone from internet.

chadf, 2021-09-02:
Could an [obvious] comment at least be added to the download page, stating something like:

We are no longer recommending the NOOBS installer. If you require multiboot, then you are SOOL^H^H^H^H^H^H^H^H^H^H^H^H^H consider using one of these other projects:

<links to Berryboot and PINN sites>

Yes, there is a little blurb about NOOBS buried in the middle of the "getting started" page, but if many(/most?) people bypass that page (e.g. via a web search), it is of limited use.

It is very irritating to waste time going around and around in circles, trying to find the download link for software that is documented [all over the place] to be there. Then waste more time trying to figure out how to do multiboot in the new imager, before finally coming across a search result that mentions NOOBS is deprecated and the new "replacement" doesn't handle multiboot. Please, please don't be like the megacorps which routinely discontinue/remove items and treat their users like dirt, expecting people to just fall in line with their latest corporate agenda product.

lurch, 2021-09-02:
> Please, please don't be like the megacorps which routinely discontinue/remove items

Raspberry Pi OS continues to run and be supported on every model of Raspberry Pi, from the original Raspberry Pi 1 Model B all the way up to the latest Raspberry Pi 400.

JamesH65, 2021-09-02:
Not only that, but we still continue to make the Pi1, Pi2 and Pi3, despite having the Pi4.

At some point though, you do need to deprecate stuff that you're no longer developing, or has been superseded by something better; in this case NOOBS, by Imager. Multiboot has always been a relatively minor use case, and that's really the only missing part, everything else is much better.

chadf, 2021-09-02:
It isn't the OS itself (in this instance) which is the issue, it's the tools and support for that OS. How many times has Microsoft or Google, for example, had a popular application, framework/toolkit, or even OS, which they one day decide to discontinue for some reason (i.e. "it isn't making them a 2000% profit; its existence discourages users from using some hyped up alternative they also have; it doesn't integrate into some proprietary architecture they want get users locked into; etc..)? Then all the users which relied on that [now discontinued] product, to do exactly what they needed, now have to scramble to work around it.

All I wanted to do was stick NOOBS on a fresh SD card and reinstalled with the latest version of Raspberry OS with a dual boot of LibreELEC.. maybe 15 minutes of my time. But instead I wasted about half an hour looking for the setup software I needed, instead of two minutes downloading NOOBS (or something else with equivalent functionality). And now I'm probably going to spend hours, maybe days more researching, installing, and testing/tweaking one of these alternative multiboot projects. All to do, effectively, what I had planned to spend about 15 minutes of my time doing.

As a user, the experience has left a bitter taste in my mouth.. so in the future, when I might consider buying more raspberry pi hardware, I will have this hesitations of whether it's worth it, wondering if I will get burned again.

JamesH65, 2021-09-02:
TBH, I've never understood the need to multiboot when SD cards are so cheap. Just swap cards. Or if you really need multiboot, use PINN which is actually better than NOOBS anyway (no dev work has been done on NOOBS for years).

As for NOOBS, a simple Google brought up this page, where is says NOOBS is no longer recommended. https://www.raspberrypi.org/documentation/computers/getting-started.html

To hesitate to buy something because an *installer* doesn't do exactly what you want seems odd, especially as it only wasted 1/2hr.  Have you actually tried Raspberry Pi Imager - you'll find it so much better. I've not used anything else since that was released. It worth noting that all our installers, OS and software are free, so any mention of profit is misguided. NOOBS has been deprecated because the newer stuff is much better, it's as simple as that. No profit motive at all, we just made something that was better.

chadf, 2021-09-02:
> At some point though, you do need to deprecate stuff that you're no longer developing, or has been superseded by something better; in this case NOOBS, by Imager. 

Of course things get deprecated, but when it is done, it really should be marked as deprecated, but not removed for some reasonable period. In this case, all of the NOOBS related documentation should have clearly stated it was deprecated. This would also allow external sites to start updating their documentation, warning users to avoid NOOBS, to avoid being bind-sided.  And only then, start removing it as an option.

> Multiboot has always been a relatively minor use case, and that's really the only missing part, everything else is much better.

Just wondering.. what metrics is this based on? Has NOOBS/RPi OS been sending back telemetry all these years, every time it boots? Honestly, I would think the exact opposite. The nice thing about RPis is that many OSes can run on it, and making it very easy for new users to try them out should be an important feature. BUT.. to try them out, a new user would either have to repeatedly wipe/reinstall their SD card [loosing any settings] or have a bunch of unused cards laying around (which I wouldn't expect most people to have just sitting around, going to waste). And even then, it requires physically swapping the card, repeatedly.. not to mention, isn't very accessibility friendly for those with physically limitations. Overall, this would only seem it discourage experimentation.

JamesH65, 2021-09-02:
The metric is that NOOBS hasn't had a link on the website for about a year, if not more (I'd need to check to be entirely accurate), but it's certainly a long time) and you are one of only a few people to have found that to be a problem.  The vast majority just use multiple SD cards, they are ridiculously cheap; it's certainly what we do in house, no-one uses NOOBS. NOOBS has never sent telemetry AFAIK. Anyway, won't be adding anything else to this conversation. If you don't want to buy Pi's because of this, that is certainly your prerogative.

efa, 2021-09-02:
I think that due to the fact that the uSD are so cheap, everyone buys a very large one and installs on it all operating systems that can

lurch, 2021-09-03:
> BUT.. to try them out, a new user would either have to repeatedly wipe/reinstall their SD card [loosing any settings]

Just as an FYI, that's also what NOOBS does - every time you add / remove / change an OS, it wipes out whatever was already on the SD card and reinstalls each OS afresh. By contrast, PINN (which as James points out has had more development done on it than NOOBS) _does_ allow you to reinstall / replace an OS without wiping out other existing OSes on the card.

happanda, 2021-09-07:
Just another user-story about wasted time hinting that there's an issue with user experience somewhere.
A couple of days ago I decided to reinstall systems on my pi's sd card: Raspberry OS + OSMC + Recalbox. First thing I've tried was Imager. An hour later after trial-and-error, searching for hidden menus (why is it even hidden by Ctrl+Shift+X?), googling, etc. I start to get the idea that Imager is not capable of multisystem installation. What led me to losing my time?
- previous installation software - NOOBS had this capability, so it felt obvious Imager would also have the feature
- there's no explicit statement about inability to multiboot anywhere on the Imager page
- NOOBS mentions have disappeared from raspberrypi website (but not completely)
- there's no links to 3rd party projects that still do multiboot
This whole situation is much more confusing than having a link to outdated software.

Also on the multibooting vs buying and using multiple sd cards:
- multiboot is one keypress away while changing sd cards is a physical process of taking out and inserting small objects into the slot. The raspberry pi itself might be hard to access because it resides somewhere on a top shelf
- with multiboot I don't have to find a place to store my collection of tiny easy-to-be-lost sd cards
- how would I mark sd cards about their contents? Do I put tiny stickers on them? Hard to believe they would hold for long
- eco-friendly multiboot: with multiboot there's no need for another piece of plastic being produced

It is a common and widespread practice for OSS to provide a link to "older versions" on their websites. It's a matter of web UX to make it non-confusing for new users.

aallan, 2021-09-07:
> NOOBS mentions have disappeared from raspberrypi website (but not completely)

Can you point me at any remaining mentions so I can remove them? Thanks!

happanda, 2021-09-07:
> Can you point me at any remaining mentions so I can remove them? Thanks!
https://www.raspberrypi.org/help/noobs%20setup/
https://projects.raspberrypi.org/en/projects/noobs-install/

aallan, 2021-09-07:
> https://www.raspberrypi.org/help/noobs%20setup/
> https://projects.raspberrypi.org/en/projects/noobs-install/

Thanks. I'll get those removed.

lurch, 2021-09-08:
>  a place to store my collection of tiny easy-to-be-lost sd cards

I use [one of these](https://www.amazon.co.uk/Resistant-Memory-Card-Case-MicroSD-MCSD16/dp/B00FAVH8NG/).

> how would I mark sd cards about their contents? Do I put tiny stickers on them? Hard to believe they would hold for long

That's what I do - I put a small sticker on each of my microSD cards with a unique number on it, and then on a separate piece of paper I write down (with pencil) which OS I've installed onto which numbered microSD card.
The other small "advantage" to separate SD cards as opposed to multiboot, is that if you totally screw something up you only lose a single OS, not all of them :laughing:

procount, 2021-09-08:
> The other small "advantage" to separate SD cards as opposed to multiboot, is that if you totally screw something up you only lose a single OS, not all of them 😆

Only applicable to NOOBS, but not other multiboot solutions like PINN or Berryboot.

efa, 2021-09-08:
since NOOBS was the standard until a few months ago, and that it proposed Raspbian and LibreELEC, everyone I know who has a Raspberry Pi has put at least those two OS's on the uSD.

Guys we are talking about linux systems, I have never reinstalled (only upgrade) a linux system in the last 10 years, I think it is practically impossible to damage a such system that you have to reinstall like you do on Windows, you just have to do it on purpose. Googling is always the quickest solution to fix. Unbelievable someone still speak about reinstall!

JamesH65, 2021-09-08:
We've not done any dev work on NOOBS for years, as Imager has completely replaced it. Imager was released March 2020, and we have not recommended NOOBS since then, if not before. 

As for reinstalling, although my Ubuntu desktop has not been reinstalled for many many years, I redo Raspberry Pi OS images all the time, not because they are damaged, just because I need to start from scratch with many things. I suspect people are often reimaging cards when moving between projects.

procount, 2021-09-08:
@JamesH65, @aallan  - to appease the die-hard NOOBS users, and anyone else who wants to multi-boot, if you want to put a small notice somewhere saying that NOOBS is deprecated and users wanting to multiboot should look towards PINN, with appropriate disclaimers that you don't endorse it etc., you're very welcome to add a link to my PINN sticky forum topic at https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574 .  

I would also welcome PINN being included in the list of "distros" that the new imaging utility can install.

aallan, 2021-09-08:
>  I suspect people are often reimaging cards when moving between projects.

If I'm using a Pi for a server application, I'll tend to update and upgrade in place. Otherwise, for most other use cases, I just reinstall. It's easier.

closed 2022-01-13 by XECDesign