raspberrypi/noobs #638: Can i install NOOBS, after installing any OS on my SD?

MochyPei, 2021-10-31:
I installed Raspbian OS on SD card, and, i really don't want to re-configure and install everything again, can i install NOOBS over Raspbian and how, or can I somehow transfer Raspbian to a pre-installed NOOBS. If I can then how, i new in linux and not used NOOBS. I hope another 16GB SD card is enough, since my Raspbian is installed on 64GB SD, and I don't even know how much it weighs. _(I hope I didn't make any mistakes, my English is still bad)_
If this need, i have Raspberry Pi 4B 8gb

lurch, 2021-10-31:
No, you can't "upgrade" a non-NOOBS installation into a NOOBS card.
The easiest option would be to keep running Raspbian on your existing SD card, and then just run NOOBS (and install other OSes except for Raspbian?) on your new card.

If you're feeling brave, you _might_ be able to "backup" your existing Raspbian installation into a NOOBS-compatible format, and then "restore" that into NOOBS, but that's probably not recommended for somebody new to Linux. I believe @procount has instructions for how to do that somewhere on his PINN wiki? (PINN is an enhanced version of NOOBS)

MochyPei, 2021-10-31:
I have already installed NOOBS, and deleted Raspbian, and I got errors with formatting the flash drive a couple of times and looked at the inscription "Waiting SD" for about an hour, I would like to clarify that my new SD card is busy with other tasks, I could only temporarily remove it and copy data from it, and another small question, is it possible to do without a mouse in NOOBS?

lurch, 2021-10-31:
NOOBS is now deprecated and not really actively developed or supported any more, so you might have more luck experimenting with https://github.com/procount/pinn ?

MochyPei, 2021-10-31:
Yeah i see, i can't install anything in NOOBS and take "Error downloading meta file"

lurch, 2021-10-31:
Yeah, some OSes now have "problems" downloading due to the expiry of an SSL certificate. PINN has added a workaround for that https://github.com/procount/pinn/issues/560

closed 2022-01-13 by XECDesign