raspberrypi/noobs #640: which Raspbian release is packed with which Noobs release?

cipy, 2021-11-24:
it would be good to state visibly here in the repo what is packaged with every single release, as the current "bullseye" is not what noobs might desire/want considering the half deprecated tools without much of a warning 😞 thanks,

lurch, 2021-11-24:
NOOBS itself is deprecated - we now recommend [Raspberry Pi Imager](https://www.raspberrypi.com/documentation/computers/getting-started.html#using-raspberry-pi-imager). If you use NOOBS with an active internet connection, then it'll automatically install the latest version of Raspberry Pi OS according to what http://downloads.raspberrypi.org/os_list_v3.json says.

procount, 2021-11-24:
As the full NOOBS will contain the most up to date version of Raspberry Pi OS at the time of release, you can go by the release date and look up the os version available at that time.

If you download full NOOBS, you can look at the version field in os.json file in the respective folder under the /os/ folder.

As NOOBS is now deprecated, you could always switch to PINN. In the next release, it will display the version information of each os. This will permit both Buster and Bullseye to be added to the repository for a transition period.

XECDesign, 2021-11-24:
> the current "bullseye" is not what noobs might desire/want considering the half deprecated tools without much of a warning

I don't quite understand what you mean there.

closed 2021-12-07 by XECDesign