raspberrypi/noobs #641: please provide noobs zips under github releases

nanderer, 2021-12-23:
its much simpler that way for people with high latency connections where you can brew a hot tea after every page click or wait 24h until a full domain is crawled. github has an public interface that can be grabbed and mailed to you by remotely running algorithms that do stuff like clone the git repo and download the 3 latest releases, all wiki pages and an overview to the project, like list of open tickets (depending on configuration) and thereby can still be operated over satellites...

thanks in advance

XECDesign, 2021-12-24:
No need to refresh pages or crawl anything. You can download the latest version of NOOBS through the following links:

https://downloads.raspberrypi.org/NOOBS_latest

https://downloads.raspberrypi.org/NOOBS_lite_latest

I'd need to make sure assets are added automatically, but since NOOBS is deprecated in favour of rpi-imager, there's not much reason to spend time on that.

closed 2021-12-24 by XECDesign

procount, 2021-12-24:
I have been adding the image version of PINN as a release asset for a while, manually. I would be interested to know how I can create a release and add an asset automatically.

lurch, 2021-12-24:
@procount Probably need to use the GitHub API? https://docs.github.com/en/rest/reference/releases

procount, 2021-12-24:
I just found this -> https://gist.github.com/stefanbuck/ce788fee19ab6eb0b4447a85fc99f447 which looks useful