raspberrypi/noobs #642: can't download lakka for rpi4

Therealsidthecat, 2021-12-27:
when i try to install lakka multiple "can't download file" errors"

XECDesign, 2021-12-27:
Fixed in https://github.com/raspberrypi/noobs/commit/6df566a76b9b893e4521313e9e0d531a99cbaba5

Should be in the  next release of NOOBS, which will be in the new year.

closed 2021-12-27 by XECDesign

lurch, 2021-12-27:
:point_up: ping @procount who I know was also having problems with SSL certificates.

procount, 2021-12-27:
I thought this was fixed.
Was a certificate missed?

lurch, 2021-12-27:
@procount Hmm, perhaps I'm getting my issues muddled up, in which case I apologise; but I thought this was the Let'sEncrypt thing where you had to turn off all SSL security in order to avoid the warning dialogs? :man_shrugging:

procount, 2021-12-27:
Already fixed in PINN, but maybe NOOBS is a bit behind....